tanakamasayuki/Arduino_TensorFlowLite_ESP32 #1: Training a model for microspeech example

Yogi3898, 2020-07-17:
Hii,

I tried the microspeech example on esp32, It is working with the default model file but when I change the model file it doesn't run, So can you let me know how to train a custom model for microspeech example.
I'm stuck in this since few weeks, So please help to solve this issue.

thank you

Yogi3898, 2020-07-17:
> Hi!
>
> This library is version 2.1.1.
>
>
> https://github.com/tensorflow/tensorflow/tree/v2.1.1/tensorflow/lite/experimental/micro/examples/micro_speech
>
> The above is the explanation page.

Hii,

This library is using an outdated version of Tensorflow
*tf-nightly-gpu==1.15.0.dev20190729*, which is removed from the source. So
I'm not able to train a model for this example.

tanakamasayuki, 2020-07-17:
https://github.com/tensorflow/tensorflow/issues/37317

Does this ticket help?

https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/examples/micro_speech/train

The latest version does not seem to support local machines.

Yogi3898, 2020-07-17:
> tensorflow/tensorflow#37317

I tried the same github repo to train the model you mentioned, but the
model produced by this script doesn't work with the example. So can we do
some changes in the library to support the current tensorflow version?

tanakamasayuki, 2020-07-17:
TensorFlow and TensorFlowLite are different products.
TensorFlow models are not available in TensorFlowLite.