tanakamasayuki/Arduino_TensorFlowLite_ESP32 #3: Didn't find op for builtin opcode 'CONV_2D' version '2'

vjsyong, 2020-11-03:
Can't perform inference on a quantized model that was trained on TF2.3. It crashes when it tries to perform a CONV2D layer inference.

tanakamasayuki, 2020-11-04:
https://www.tensorflow.org/lite/microcontrollers/build_convert
Have you converted to microcontrollers?

https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/examples/hello_world#deploy-to-esp32
The version of the Arduino is old and the latest needs to be built with ESP-IDF4.

amanglarus, 2022-10-13:
hey i am also getting same error i am using tf 2.8.0 can you let me know how can resolve this error

amanglarus, 2022-10-13:
> https://www.tensorflow.org/lite/microcontrollers/build_convert Have you converted to microcontrollers?
> 
> https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/examples/hello_world#deploy-to-esp32 The version of the Arduino is old and the latest needs to be built with ESP-IDF4.

hey please resolve my problem