tanakamasayuki/Arduino_TensorFlowLite_ESP32 #5: Old version, missing opcodes

tcirstea, 2020-12-11:
Hi, it seems like the version of this library is old and out of sync with the official libraries.
Running a model with a few dense layers, the first one does a sub operation and it is missing.

```
Didn't find op for builtin opcode 'SUB' version '1'
Failed to get registration from op code  d
```

Looking into AllOpsResolver, there's no sub operation there, but it is present in the tensorflow github (AddSub)

tcirstea, 2020-12-12:
I see the issues....

The most recent RC 2.4.0 doesn't make a build on esp32 because of bugs. They seem to have added esp32 to CI so that shouldn't happen again.

Previous versions higher than yours worked (i.e. the opcode SUB was present), but only when building with arduino. When building with espidf, there would be a byte alignment issue (8 bytes lost due to alignment. To avoid this loss, please make sure the tensor_arena is 16 bytes aligned.)
The byte alignment was fixed in september but due to the lack of a functional build, I'm using nightly as the most recent.

I ended up forking it for my own use https://github.com/tcirstea/Arduino_TensorFlowLite_ESP32
Note if you want to pull those changes in, I removed the examples (since import locations and namespaces changed), I changed the readme and changed the library.properties.

I have both v2.3.1 for arduino and v2.4.0-rc-nightly for espidf on there.
Feel free to close this issue if you like.

tanakamasayuki, 2020-12-14:
Thank you very much.
I would like to update it once it is officially released.