tanakamasayuki/Arduino_TensorFlowLite_ESP32 #6: trained model using official tensorflow lite microspeach example is incompatible with the current micro speach example for ESP32

hangman1998, 2021-05-27:
Hi.
I have trained a network using the following colab notebook <https://colab.research.google.com/github/tensorflow/tensorflow/blob/master/tensorflow/lite/micro/examples/micro_speech/train/train_micro_speech_model.ipynb>.
Now following the instructions explained in **Deploy to ESP32** section of the official TensorFlow lite microspeech example (<https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/examples/micro_speech#deploy-to-esp32>) and using the latest version of ESP IDF I receive errors regarding the difference between the input shape of the model and the expected input shape.
After viewing the already used model in the microspeech example in netron and contrasting it to the trained model I realized that one has input shape of (-1, 1960) while the other has input shape of (-1,49,40,1).
Furthermore one accepts inputs of type UINT8 while the other expects inputs to be of type INT8.

 
[image]
[image]

if updating the repo to be compatible with the latest trained model is not currently possible, it would help me a lot to know exactly what versions of TensorFlow are compatible with the current state of this repo, and training my network using them would not cause these incompatibilities.
best regards,
Yousef

tanakamasayuki, 2021-05-28:
It is based on TFLite 2.1.1.
It is old and does not support new formats.
The official Arduino library version is 2.4.
GitHub seems to have 2.5.

I hope to update it soon.

Robertou2, 2021-05-28:
@hangman1998  Maybe this tutorial help you  https://www.survivingwithandroid.com/tensorflow-lite-micro-esp32-cam-fashion-mnist-platformio/

AlessioGagliardi, 2021-06-14:
> Hi.
> I have trained a network using the following colab notebook [...]

Same issues, have you find a way to build the new lib? I started from the tf github example, copied the new lib, canged from .cc to .cpp e inserted into the proper arduino lib folder. If i try to compile the "micro speech" tf example i run into this error: "'to_string' is not a member of 'std'".

Probably is the c++11 compiler. Anyone have fond a way to solve the issue? Thanks in advance.

Ale

tanakamasayuki, 2021-06-15:
The Arduino version is old.
Please try using the native version of ESP-IDF.

https://www.tensorflow.org/lite/microcontrollers
https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/examples/hello_world#deploy-to-esp32

tanakamasayuki, 2021-10-05:
The Arduino version is in the works.

https://www.tensorflow.org/lite/microcontrollers
The official ESP-IDF version is recommended.

Reyeseyer, 2022-01-26:
I have your version running on my ESP32 under Arduino ESP32 SDK but I have tried to train my own  words and I get the error:

Didn't find op for builtin opcode 'RESHAPE' version '1'

Failed to get registration from op code  d
 
AllocateTensors() failed

So I tried to train the speech example (https://colab.research.google.com/github/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/micro_speech/train/train_micro_speech_model.ipynb ) with the same result, neural network is almost the same than yours with very close length (18208 vs 18712). So I think that maybe it is a version error or something in the example, can you please upload your train code so I can test it? Because I think that if I can test a neural network and make it work like yours I could find the error in my own.

As adiccional material, I let you the comparative of Netron schemas of tflite networks:
Example schema in github of tflite:
[image]

My trained net from example:
[image]