tanakamasayuki/Arduino_TensorFlowLite_ESP32 #7: The micro_speech_ESP_EYE example does not work correctly.

zeynepdicle, 2021-10-04:
I have an Esp32 Wrover-e board with MP34DT05 microphone. I am running the micro_speech example but I am not getting the correct result. It always prints "no" even though it is not said.
What do I need to change to get the correct result?

tanakamasayuki, 2021-10-04:
It's difficult to recognize.
A new version is out and is in preparation.

Tip. You need to pronounce the words for recognition.
It is difficult to pronounce in a normal way.

zeynepdicle, 2021-10-04:
Yes, I know I have to pronounce words to be recognized.
I don't get the same issue for "yes".

tanakamasayuki, 2021-10-04:
I think the preset model is bad.
The accuracy is not high.
However, since it is an old version, it may be improved if it is the latest version.

zeynepdicle, 2021-10-04:
When will the new version be released?

tanakamasayuki, 2021-10-05:
The Arduino version is in the works.

https://www.tensorflow.org/lite/microcontrollers
The official ESP-IDF version is recommended.

closed 2023-03-01 by zeynepdicle