tanakamasayuki/Arduino_TensorFlowLite_ESP32 #10: Support for LSTM and Autoencoder

expeon07, 2022-03-23:
Hi, I'm trying to implement a multivariate time series prediction model using LSTM, and an anomaly detection model using autoencoders. I have successfully converted them to a TFLite version and tried to upload them in my ESP32.

However I get the following error:
`Only 1 subgraph is currently supported.`

Are LSTMs and anomaly detection supported by this library? If so, are there special configurations that I need to do?

Thanks!

koulnodji6056, 2022-12-02:
@expeon07 could you please share with me the code for converting the LSTM model to TFLite?
I have not succeeded in the conversion. Thank you

expeon07, 2022-12-07:
> @expeon07 could you please share with me the code for converting the LSTM model to TFLite? I have not succeeded in the conversion. Thank you

Hi, I unrolled the LSTM layer by turning unroll=True (default to false)

koulnodji6056, 2022-12-10:
@expeon07  please I am getting it clear. 
Is that turning of  unroll=True (default to false) was done during model training of during model conversion (TFliteConverter)?
kindly, I request more explanation or a close orientation. 
NB: I don't have much experience in coding

expeon07, 2022-12-10:
> @expeon07 please I am not getting it clear. Is that turning of unroll=True (default to false) was done during model training of during model conversion (TFliteConverter)? kindly, I request more explanation or a close orientation. NB: I don't have much experience in coding

Hi, I think it's during the model training. I cannot remember as I only tried it months ago. Have you tried doing either/both yourself? You can ask the developers in the TFlite repo as well

koulnodji6056, 2022-12-10:
@expeon07 Noted with thanks