tanakamasayuki/Arduino_TensorFlowLite_ESP32 #11: TensorFlowLite_ESP32\src\bus\i2s_lcd_esp32_driver.c:29:25: fatal error: hal/gpio_ll.h: No such file or directory

phuzybuny, 2022-08-30:
Was initially encountering the error `TensorFlowLite_ESP32\src\bus\i2s_lcd_esp32_driver.c:24:30: fatal error: esp32/rom/lldesc.h: No such file or directory`. But changing the `#include "esp32/rom/lldesc.h"` to `#include "rom/lldesc.h"` worked on my Windows machine with the Espressif ESP32 boards definition version 1.0.6.

However, I could not locate the equivalent for `hal/gpio_ll.h`.

theloukou, 2022-11-11:
Run into the same thing today.
TFLite_ESP32 V1.0.0 seems to need ESP32 platform V2.0.0 or above to function properly.
I wonder if it is possible to make this as a requirement, so it wont even try to compile without the proper platform version.

xiaoyinliuyun, 2023-04-21:
Started compiling on arduino ran into this problem, but did not run into this problem on vscode.

xiaoyinliuyun, 2023-04-21:
开始在arduino上编译遇到这个问题，但是在vscode上没有遇到这个问题。

theloukou, 2023-04-22:
I ended up using raw TFlite. A bit more involved, but made my own wrappers to speed things up.
I consider this project dead after all this inactivity tbh.

geek394, 2023-07-15:
> 我最终使用了原始TFlite。稍微复杂一点，但制作了我自己的包装器来加快速度。我认为这个项目在所有这些不活动之后已经死了 tbh。

The original TFlite refers to version 0.9.0

KyleQ1, 2024-07-15:
I used platformio to fix this.