tanakamasayuki/Arduino_TensorFlowLite_ESP32 #22: I installed the last version of this library and when I tried the micro_speech in Arduino IDE it did not compile.

abmj01, 2024-11-27:
Hi Everyone,

I am a student that is currently trying to do an AI voice command recognition with a i2s microphone(INMP441) and an esp32-s3- wroom. And I have come to your amazing library that should help me with that.

1) I installed your library via Arduino IDE.
2) In Arduino IDE I navigated to File-> examples -> TensorFlowLite_ESP32 -> micro_speech and opened the example.
3) I changed some values or parameters in the audio_provider.cpp to match the pins of the ESP32 I am using and already tested. 

When I compiled the program I got the following errors, can you guide me of what should I do? Should I use another program such as platformio? :

c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src\tensorflow\lite\micro\micro_error_reporter.cpp: In function 'tflite::ErrorReporter* tflite::GetMicroErrorReporter()':
c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src\tensorflow\lite\micro\micro_error_reporter.cpp:58:76: error: 'static void tflite::MicroErrorReporter::operator delete(void*)' is private within this context
   58 |     error_reporter_ = new (micro_error_reporter_buffer) MicroErrorReporter();
      |                                                                            ^
In file included from c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src/tensorflow/lite/micro/micro_error_reporter.h:21,
                 from c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src\tensorflow\lite\micro\micro_error_reporter.cpp:16:
c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src/tensorflow/lite/micro/compatibility.h:27:8: note: declared private here
   27 |   void operator delete(void* p) {}
      |        ^~~~~~~~
c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src/tensorflow/lite/micro/micro_error_reporter.h:51:3: note: in expansion of macro 'TF_LITE_REMOVE_VIRTUAL_DELETE'
   51 |   TF_LITE_REMOVE_VIRTUAL_DELETE
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src\tensorflow\lite\micro\micro_allocator.cpp: In static member function 'static tflite::MicroAllocator* tflite::MicroAllocator::Create(uint8_t*, size_t, tflite::ErrorReporter*)':
c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src\tensorflow\lite\micro\micro_allocator.cpp:354:55: error: 'static void tflite::GreedyMemoryPlanner::operator delete(void*)' is private within this context
  354 |       new (memory_planner_buffer) GreedyMemoryPlanner();
      |                                                       ^
In file included from c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src/tensorflow/lite/micro/arena_allocator/simple_memory_allocator.h:25,
                 from c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src/tensorflow/lite/micro/micro_allocator.h:24,
                 from c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src\tensorflow\lite\micro\micro_allocator.cpp:16:
c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src/tensorflow/lite/micro/compatibility.h:27:8: note: declared private here
   27 |   void operator delete(void* p) {}
      |        ^~~~~~~~
c:\Users\ahmed\Documents\Arduino\libraries\TensorFlowLite_ESP32\src/tensorflow/lite/micro/memory_planner/greedy_memory_planner.h:162:3: note: in expansion of macro 'TF_LITE_REMOVE_VIRTUAL_DELETE'
  162 |   TF_LITE_REMOVE_VIRTUAL_DELETE
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Thanks in advance for your cooperation and understanding.

Yassy03, 2024-11-28:
i got the same error !

jhohannfabian, 2025-01-15:
I got the same error!!

yanho824, 2025-07-03:
I have the same problem, did you solve it?

closed 2026-07-03 by tanakamasayuki